TheTorProject/lepidopter #10: Plug and Play

mrphs, 2015-10-26:
Please make sure everything is plug and play. The operator shouldn't necessarily need to have a keyboard or monitor. Everything should be running smoothly as soon as it's powered up and plugged into ethernet.

hellais, 2015-10-27:
I believe this is already the case @anadahz can you confirm?

hellais, 2015-10-27:
@mrphs can you tag issues in the future?

mrphs, 2015-10-27:
@hellais hmm sure... I just don't think I've the ability to do so. of if I do, I'm not sure how :)

hellais, 2015-10-27:
You should now if you accept the invite to join the OONI group.

anadahz, 2015-10-28:
Lepidopter already does this.
You can check all the relevant scheduled commands (cronjobs):
── cron.daily
│   │   │   ├── run_ooniprobe_deck
│   │   │   └── upload_reports
│   │   ├── cron.weekly
│   │   │   ├── remove_inc_reports
│   │   │   ├── update_deck
│   │   │   └── update_ooniprobe
Excerpt from: https://github.com/TheTorProject/lepidopter#lepidopter-repository-tree-structure

closed 2015-10-28 by anadahz