TheTorProject/lepidopter #26: Publish a working image on the web

hellais, 2016-04-29:
This means making a tested build of the raspberry pi image somewhere on the web and pointing to it's download location on the lepidopter page.

anadahz, 2016-05-08:
What would be the best way to ensure the authenticity of this image hosted online?
Shall we create an offline key used to sign the images?

hellais, 2016-05-08:
I would say you sign it with your personal key for the moment. Maybe in the future when we will have in place some sort of build system we can consider having a shared key for signing the images.

anadahz, 2016-05-23:
The image is out.
Updated [instructions](https://github.com/TheTorProject/ooni-web/pull/53) or in OONI [website](https://ooni.torproject.org/lepidopter/lepidopter/) when the changes are merged.

closed 2016-05-23 by anadahz