TheTorProject/lepidopter #35: Add support to configuring a tor hidden service for SSH access

anadahz, 2016-06-03:
The build script should also support configuring a tor hidden service that allows ssh access to some set of keys.

anadahz, 2016-08-06:
Prior to enabling access to a set of ssh keys we 'll first need to set the group/user permissions in lepidopter. Currently the default username (lepidopter) is on the sudo group and requires user authentication (default password: lepidopter) to run sudo.
In order to access lepidopter via ssh and perform superuser tasks (with sudo) we 'll either need to know the password of the user, disable sudo user authentication or log as root via ssh.

If we expose the ssh service via a tor hidden service we should ensure that the default authentication password is being changed or allow only public key authentication (which may not be applicable to headless lepidopter setups).

anadahz, 2016-08-28:
@hellais, @darkk, @bassosimone any thoughts on https://github.com/TheTorProject/lepidopter/issues/35#issuecomment-238049020?

bassosimone, 2016-08-29:
@anadahz, thanks for the question, I guess that's a very important one. So, let's discuss this and here's what I just said in the IRC meeting about this issue:

> if possible I'd avoid us having root access on lepidopters because it increases the scope of what we can do using the probes way beyond the software we deploy using standard channels and this imho could put partners in a more troubling situation if cought, not to mention that say I have access to all lepidopters, I am compromised, and someone uses that access to do nasty things

anadahz, 2016-08-30:
After last meeting's discussion it seems that we are going to drop the idea of using a tor HS for SSH access in lepidopter images.

hellais, 2016-08-31:
I would say we keep this as a ticket, but defer it to future versions.