TheTorProject/lepidopter #37: Implement a setup wizard

anadahz, 2016-06-03:
This will be displayed only the first time the probe is setup. It will guide the user through configuring the following setting from a web interface:

- [ ] The WPA password and password for accessing the admin interface (perhaps they could both be the same?)
- [ ] If the probe should also expose a tor hidden service that grants OONI team remote access.
- [ ] If the Web GUI should also be exposed as a Tor Hidden Service (should this HS address be the same as the one above?).
- [ ] What types of measurements should be run with a description of each test and their associated inputs.
- [ ] When the wizard is complete it will provide a summary of the configured options. 
- [ ] If the user has chosen to enable Tor Hidden Service remote administration it will also print out the tor hidden service address and the list of authorized keys. The user may add their own key or remove some keys from the list.
- [ ] When the users click save they will be redirected to the Web GUI.

willscott, 2016-06-06:
The default password for the device should also be changed from 'lepidopter' as part of initial setup.

anadahz, 2016-06-06:
@willscott I update the setup wizard requirements to explicitly mention the change of the default system password.

anadahz, 2016-10-23:
I had a look on how much work is going to be needed to implement this task.
There is a WiFi configuration portal WUI [Raspap-webgui] (https://github.com/billz/raspap-webgui) that already includes a number of the configuration options/services that we would like to have implemented in https://github.com/TheTorProject/lepidopter/issues/37#issue-158380913 .

We will still need to find a way on how to integrate Raspap with ooniprobe's
WUI, implement a setup wizard and add the missing option to the Raspap WUI
interface.

Raspberry Pi 3 wlan module seems to be able handling hostapd.