TheTorProject/lepidopter #38: Lepidodpter Web GUI

anadahz, 2016-06-03:
This will be shown every time a user logs into the Lepidopter Web GUI.
It will expose the following functionality:
- [ ] List of measurements run so far with time-stamps and test type.
- [ ] The HS address of ssh and/or web GUI.
- [ ] The list of authorized ssh keys.

anadahz, 2016-10-23:
The web GUI can be an extension or perhaps even merged in #37.

koushikcs562, 2025-11-18:
Hello @anadahz,
I'm interested in working on the implementation of the main Lepidopter Web GUI. This interface is essential for ongoing system administration and viewing measurement results.
My Approach:
I agree with the idea of potentially merging or tightly integrating the Web GUI with the Setup Wizard implementation (Issue #37). The Web GUI would essentially be the same application interface, but without the mandatory setup logic blocking access after the first boot.
Core Implementation Focus
Technology Stack: I will aim to use a lightweight, modern web framework (like Flask/Django for Python or a similar framework if preferred) to ensure maintainability and a clean, responsive front-end.
Measurements Dashboard: Implement the view to List measurements run so far, providing filtering/sorting options and links to detailed results if available. This requires interfacing cleanly with the OONI Probe data storage.
HS Status Panel: Create a dedicated panel to display the status and addresses of the Tor Hidden Services (for SSH and/or the Web GUI) and manage the list of authorized SSH keys. This provides crucial security information to the user.
My goal is to create a clean, functional dashboard that meets all the listed requirements and serves as the primary control center for the Lepidopter device