TheTorProject/lepidopter #43: include wifi drivers for rpi 3

willscott, 2016-06-06:
currently, the only interface available on boot is eth0, even though the rpi 3 includes an on-board wifi chip

willscott, 2016-06-06:
it looks like it may be trying to load the brcmfmac kernel module, but it appears to fail.

relevant demsg lines:
```
systemd[1]: Failed to start Load Kernel Modules
...
brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
brcmfmac_sdio mmc1:0001:1: Direct firmware load for mrcm/mrcmfmac43430-sdio.bin failed with error -2
brcmfmac: brcmf_sdio_htclk: HT Avail timeout (10000000): clkctl 0x50
```

anadahz, 2016-06-06:
I will try to add WiFi support in next release of lepidopter.
Thank you for reporting this.

anadahz, 2016-06-29:
I need a Raspberry Pi 3 device to test this properly.

willscott, 2016-06-29:
I'll bring one to you next week.

anadahz, 2016-06-30:
@willscot excellent thanks!

PR #66 (2016-09-06): Feature/rpi3 wifi support
Fixes #43
closed 2016-09-06 by anadahz