TheTorProject/lepidopter #46: Add support for mDNS based service discovery of the address of a pi

hellais, 2016-06-11:
Currently it's hard to learn what the address of a raspberry pi based probe and it would be nice to support some mDNS based announcement and discovery mechanism.

Bonjour or Avahi come to mind and it seems like there is a pretty good python implementation of this: https://github.com/jstasiak/python-zeroconf.

I think this feature should be something that goes into ooniprobe itself (the ooniprobe daemon actually) and enabled only on raspberry pi images of it.

anadahz, 2016-06-11:
I see mDNS useful, however from my experience with avahi i ended up having many issues and not being able to detect/connect with the host in question.
We should consider also that this will not possible if we ever want to reduce the existence of lepidopter in a network.
Normally home routers have some kind of DNS setup so accessing lepidopter will be quite trivial.
You can try by pinging lepidopter, in my case with the default settings of a home router:
```
$ ping lepidopter
PING lepidopter.lan (192.168.1.3) 56(84) bytes of data.
64 bytes from lepidopter.lan (192.168.1.3): icmp_seq=1 ttl=64 time=1.28
```

irl, 2016-06-11:
On 11/06/16 16:35, anadahz wrote:
> I see mDNS useful, however from my experience with avahi i ended up
> having many issues and not being able to detect/connect with the host in
> question.
> We should consider also that this will not [...]

If lepidopter is doing DHCP requests with the hostname set to lepidopter
anyway, I don't think this is increasing the network operators
visibility greatly beyond what they can currently see.

anadahz, 2016-06-11:
@irl AFAIK there are some ways to "hide" the presense of DHCP messages in a network via some kind of covert channel. However this is not that possible with mDNS.

hellais, 2016-06-12:
Well it can perhaps be something that is optional, but I think that if we expect the user to login to the raspberry pi  there needs to be some robust discovery mechanism that should not be dependent on the DNS resolver of the router configuring entries for the device.

Not doing so makes the administering of the device not very usable.
Moreover we are currently make not effort whatsoever to "hide the presence" of a raspberry pi on the network. Doing so it also something that is actually quite difficult (I mean in the end it needs to bind on some port on the local network to expose the user interface so even if DHCP messages are "hidden" (not also sure how you would do this or what you mean by this), a port scan would still show the existence of it).
I think this is a separate issue from that of the discovery of the device (which we should support by default) and should be discussed in the context of another ticket.

@anadahz what sort of issues were you having with avahi? I have not used avahi, but have played around a bit with Bonjour and it does work as expected.
In the end it's a pretty straightforward protocol, it just sends a DNS query using multicast UDP towards port 5353 and when the device receives the query it sends a response with in the payload containing the local address of the device.

anadahz, 2016-06-12:
@hellais while i like the idea of having zeroconf to ease the configuration of lepidopter at the current state i think is not that important. I think a person that can connect to a network device via SSH, control and administer this device can almost certain be able to find the IP address or connect to lepidopter.
I guess it makes more sense to re-evaluate the mDNS/zeroconf for the lepidopter WiFi version.

I believe going into details on how one can secure a network to disallow port scanning, limit certain devices in one VLAN and hide DHCP messages or how one can misuse mDNS to initiate DoS attacks are out of context here.

hellais, 2016-06-12:
> @hellais while i like the idea of having zeroconf to ease the configuration of lepidopter at the current state i think is not that important. I think a person that can connect to a network device [...]

Yup I agree on this point.

> I guess it makes more sense to re-evaluate the mDNS/zeroconf for the lepidopter WiFi version.

Well I am thinking this should be a requirement even before we have the WiFi version, but also as soon as we have the web GUI for ooniprobe (that is going to be part of the beta release of lepidopter).

> I believe going into details on how one can secure a network to disallow port scanning, limit certain devices in one VLAN and hide DHCP messages or how one can misuse mDNS to initiate DoS attacks are out of context here.

Yes I agree that these are out of scope here.

anadahz, 2016-08-30:
It seems that `avahi-daemon` will be the best option to implement Bonjour/mDNS based service discovery.

I'm planning to add the following services that will be advertised on the network:
- [ ]  ooniprobe web service
- [ ] SSH

hellais, 2016-08-30:
Excellent!

Don't forget to write these also as an update script so we can support also legacy lepidopters.

darkk, 2016-09-06:
We may also fallback to IPv6 local address if mDNS is broken.
Also, do we support two lepidopters running in same LAN? If we don't, we can use some well-known hardcoded IPv6 address in `fe80::/64` address space.

For example, mDNS is turned off in Ubuntu when `*.local` hostnames are resolvable via usual DNS. It may happen when ISP wants to provide `retracker.local` services to handle bittorrent traffic.

anadahz, 2016-09-13:
> We may also fallback to IPv6 local address if mDNS is broken.
I guess it make more sense to use IPv6 when OONI has support for it.

> Also, do we support two lepidopters running in same LAN?
By default the avahi service will append a digit (-n) on the hostname in case of duplicate hostnames.

anadahz, 2016-09-14:
I'm going to close this as it has been implement feel free to open a new ticket regarding IPv6 support.

closed 2016-09-14 by anadahz