TheTorProject/lepidopter #50: Reduce SD card wear out

anadahz, 2016-06-11:
Evaluate options to reduce writes in order to minimize SD card wear out.
One idea is to move all logs files in RAM and save before shutdown/reboot.
In a case of an unexpected shutdown (power disruption) no logs files will be
saved, though SD card longevity is way more important!

darkk, 2016-08-18:
We should probably estimate number of writes we have and number of writes we're targeting.

Anecdotal evidence is: I've been running Tor relay on RPi 24/7 and it had not killed SD card within a year, but PSU died.

anadahz, 2017-04-24:
Currently the best options in order to extend the life of an SD card are the
following:
* [x] Disable the swap partition
* [ ] Enable noatime mount option
* [x] Free up as much space on the SD card as possible (#22, #14)

Further improvements that are not recommended in the current version of
lepidopter, mainly due to power loss and no availability of an external disk:
* Disable the filesystem journal
* Specific directories (the one with high writes, eg: `/var/log`) in RAM
* Use an external hard disk or USB disk for specific directories

anadahz, 2017-05-05:
This has been implemented in: https://github.com/TheTorProject/lepidopter/releases/tag/v1.0.0

closed 2017-05-05 by anadahz