TheTorProject/lepidopter #52: Uniform the times in which period tasks get run

hellais, 2016-06-14:
Currently lepidopter uses for the period tasks (ooniprobe update, ooniprobe test runs, etc.) the `@daily`, `@weekly` cronjobs.

It's not clear to me exactly when these occur and it makes it harder to troubleshoot why a particular probe may not be submitting reports (i.e. is it due to the fact that the cronjob has not run yet or is it offline).

It would be better if we used UTC time for all the lepidopter probes for the daily measurements (so that they all run measurements at the same time, for example at 24:00 UTC) and that the weekly tasks always happen on the same day of the week.

This would also allow us to do better planning for when to issue a new release and ensure the raspberry pi images get updated.

anadahz, 2016-06-14:
Lepidopter uses (by default) the UTC time.

According to Lepidopter's (default) crontab config:
Daily tasks start at: 6:25
Weekly tasks start at: 6:47 every Sunday

anadahz, 2016-06-29:
#54 introduces a crontab file, I have given enough time for the ooniprobe and deck weekly updates to finish prior running the daily ooniprobe task.

@hellais does this cover the bug report?

hellais, 2016-06-29:
Is the timezone guaranteed to always be UTC?

anadahz, 2016-06-29:
@hellais this https://github.com/TheTorProject/lepidopter/commit/e9d99af2026e7991b729447b14d8e985bae93ed0 ensures that UTC is being used.

closed 2016-07-10 by hellais