TheTorProject/lepidopter #57: Add support for automatic updates in lepidopter

hellais, 2016-07-10:
There should be a way to automatically update the image of lepidopter.

Currently the only thing that is being updated is ooniprobe itself and this is done via pip.

The update system should support doing more advanced things as well such as:

* Updating configuration files

* Making changes to the period scripts that get run

* Installing additional software that is needed for newer version of lepidopter/ooniprobe

* Configuring ooniprobe to run in different ways

hellais, 2016-08-03:
Other things that come to mind that will be necessary include:

* Setting up a init script for running certain system services (ooniprobe-agent)

* Configuring a wifi network and altering other network settings

Moreover if we want to be ambitious it would be ideal to also support:

* Upgrading the firmware

* Upgrading the OS

anadahz, 2016-08-12:
Categorizing the requirements for the automatic update process:
- [ ] Update system/ooniprobe configuration files (this includes altering the period scripts get run and configuration of ooniprobe to run in different ways, set up an init script for running certain system services)
- [ ] Add/remove software

> Configuring a wifi network and altering other network settings

Modifying network configuration of a running (24/7) headless embedded server is usually not a good idea.
@hellais  can you elaborate a bit more of how did you think of it as something that could be useful?

> Upgrading the firmware

This is quite trivial to be done but given the nature of lepidopter distribution being a headless server we will not gain much by regularly upgrading the firmware, instead if something goes wrong during the process or after rebooting the RasPi for the firmware updates to take effect we can end up with a non-bootable or unresponsive RasPi.

> Upgrading the OS

Lepidopter uses Debian Jessie that is going to be fully supported up to [~June 6th 2018](https://www.debian.org/security/faq#lifespan) and Debian has planned to offer long term support from [May 2018 to end of April/May 2020](https://wiki.debian.org/LTS#Debian_Long_Term_Support).

Given the fact that we are going to have the ability to add/remove software in future versions of lepidopter we can still trigger (upon extensive testing) a distribution (OS) upgrade.

Does this makes sense?

darkk, 2016-08-18:
xref: #69

anadahz, 2016-08-28:
Given that OTA upgrades is a complex implementation we will need to dedicate more time for testing and deployment, it makes sense to move this ticket in later version of Lepidopter.

hellais, 2016-12-05:
I think we can consider this at this point closed?

hellais, 2017-04-05:
I am going to say this is done with lepidopter-updater

closed 2017-04-05 by hellais