TheTorProject/lepidopter #58: Support identifying the version of a lepidopter image

hellais, 2016-07-10:
Currently the lepidopter image ships with a file called `/etc/default/lepidopter` that contains some version information, though it appears to only mention the label of the version (ex. `LEPIDOPTER_BUILD="alpha"`) and not the actual full version string.

This should be a problem if we decide to make some future major releases that are also going to be in alpha.

I would also advise we change the format of the version to be something that is easier to be parsed (for example just including the raw version number) and perhaps place it inside of a more standard file (such as `/etc/lepidopter_version` similarly to how debian does it).

For backward compatibility reasons I would say we keep both files `/etc/default/lepidopter` being the variable and `/etc/lepidopter_version` being the raw version number.

anadahz, 2016-07-14:
The full version string has been introduced already and being generated upon [image build](https://github.com/TheTorProject/lepidopter/blob/master/lepidopter-vmdebootstrap_build.sh#L3)

Using a `lepidopter_version` file with the raw version number is a useful thing to have.
@hellais is `git describe --abbrev=0` output a good match for that?

hellais, 2016-07-14:
Yeah that would work, though maybe we should be also committing this to the repository to make it clear that is the location where you can find it and how it looks like.

anadahz, 2016-07-19:
@hellais what do you mean by: `committing this to the repository` ?

hellais, 2016-07-19:
I mean there should be a file inside of `https://github.com/TheTorProject/lepidopter/tree/master/lepidopter-fh/etc/default` called `lepidopter_version` that contains the above mentioned string.

anadahz, 2016-07-19:
Is there a way to do this automatically in Github?
Otherwise every person that submits a PR need to change the version of the file?

I prefer to have the version generated when the image is being build unless there is a good reason to always generate the `lepidopter_version` on every commit.

hellais, 2016-07-19:
If I understand correctly how you are setting it (based on the output of `git describe --abbrev=0`) it only needs to be updated every time you tag a new release.

anadahz, 2016-07-19:
Currently `git describe --tags` is being [used](https://github.com/TheTorProject/lepidopter/blob/master/lepidopter-vmdebootstrap_build.sh#L3).
`git describe --abbrev=0` will be introduced in the future release. I still don't understand the usefulness of it being on a file in the git repository that already show the version of the files.

Manually generating this file upon every new release can increase the probability of failure in case one will accidentally forget to include this in the current version and the `lepidopter_version` file will have the older version instead.

hellais, 2016-07-19:
It just makes it more clear to somebody that wants to use this file that it's present on the filesystem.

It's also a pretty standard thing to do to have the version information be committed inside of the code repository and not have to depend on external tools (such a git to obtain it).

anadahz, 2016-09-14:
Implemented in #71.

PR #71 (2016-09-14): Add lepidopter release versioning script, rename Changelog
This fixes #58.
closed 2016-09-14 by anadahz