TheTorProject/lepidopter #59: Add annotation for report submitted by partner probes

hellais, 2016-07-10:
There should be some programmatic easy way of having an optional probe_id included as an annotation inside of the reports that are submitted by partners.

I would say we can also do the configuration of this as part of the testing and configuration of the probes specifically for partners and doesn't have to be shipped in the mainline distribution.

However there should be some file called something like `/etc/ooniprobe/probe_id` that when present and contains a name it will be added as an annotation `{"probe_id": "something"}`

anadahz, 2016-08-06:
@hellais Wouldn't be cleaner to add the probe_id with ooniprobe annotations parameter instead?

hellais, 2016-08-07:
I think in general passing configuration options as command line arguments is not very good approach. It also means that when ooniprobe-agent is released this will no longer be applicable.

anadahz, 2016-08-07:
Since this is going to change in ooniprobe-agent makes more sense to have the `probe_id` defined in `ooniprobe.conf` this way other people that are not submiting reports via lepidopter can set annotations. Does this makes sense?

hellais, 2016-08-08:
I think it should also be separate from the configuration file given the fact that we don't want the probe_id to be something that is set by the user, but rather automatically generated by the software itself.

We also want to have the ability to allow the user to re-generate their probe_id value and if we put this inside of the configuration file it's not going to be possible to change just that attribute without loosing comments.

That is why I am suggesting we either put it inside of `/etc/` or inside of `/var/lib/ooni`

anadahz, 2016-08-28:
Quoting @hellais in IRC: #59 is https://github.com/TheTorProject/ooni-probe/issues/438 and I think we will not include it in the short term.

@hellais is this still needed to be implemented in lepidopter?

hellais, 2016-08-31:
Yes this should be part of the probe authorship system and should not be implemented as a lepidopter specific feature.

closed 2016-08-31 by hellais