TheTorProject/lepidopter #60: Have some mechanism for assessing which partner probes are working

hellais, 2016-07-10:
Currently we have no way of learning if a probe from a partner is working or not and we have to rely on waiting for them to perform the first measurement to know if they are even online.

I would say as a first easy thing to support we could have a service that we ping as soon as the probe comes online (and perhaps have it include the probe_id if present) so that at least we know which ones are able to connect to the internet.

For added stealth we could do this either via Tor or cloudfronting.

anadahz, 2016-08-30:
In order to avoid maintaining more services does it makes more sense to send an encrypted email notification upon successful bootstrap of the image to an email address that we control.?

We need to make sure that we check for a file that will indicate that the probe is a partner probe.

hellais, 2016-08-31:
I started adding hooks to support this inside of ooniprobe itself. I don't expect this to be done as part of the 2.0.0, but it should be included in future versions. 

Migrated this to: https://github.com/TheTorProject/ooni-probe/issues/596

closed 2016-08-31 by hellais