TheTorProject/lepidopter #63: Potential improvements for lepidopter

hellais, 2016-07-13:
I was just looking at what resin.io are doing for managing big fleets of embedded devices and it seems like there are some things we can learn from them.

https://resin.io/how-it-works/

Basically they have a very lightweight base OS image https://github.com/resin-os that is based on yocto (http://www.yoctoproject.org/). Then everything is an app that runs inside of a docker container. When they need to provision updates, what they do is they push a new docker image to the docker repositories and then pull it in from the devices.

The devices are managed via a VPN that allows them to reach devices behind NAT. 

In particular they have a supervisor that is written in go and based on docker for running "apps" on their devices:
* https://github.com/resin-io/resin-supervisor

A captive portal like thing for connecting devices to wifi networks:
* https://github.com/resin-io/resin-wifi-connect

bassosimone, 2016-07-13:
Nice! I think one way to address updates *could* be having the core functionality packaged all together into a Docker image, such that for us updating is just swapping images.

In turn, this would enable us to choose someone such as resin.io for managing devices.

I am now wondering if it would be technically feasible to have all the packaging inside of a docker image, or whether there are Raspberry specific files and dirs that Lepidopter needs to access.

anadahz, 2016-08-12:
I had a look at resin-os, yoctoproject and [mender](https://mender.io/); an over-the-air software updater for connected Linux devices. It seems that these projects had already solved a number of the issues and burdens that we are going to face in the future.
It's definitely worth the process to test and find out if/how any of these components and software could help us to better deploy and manage probes especially if we want to support a diverse set of hardware devices.

anadahz, 2016-08-13:
After some research it seems that there is a number of software projects that are performing OTA in various ways. The OSTree related projects page has an excellent for most of this software:
https://ostree.readthedocs.io/en/latest/manual/related-projects/

anadahz, 2016-08-28:
Adding the evaluation (WIP) spec #69 for OTA upgrades to our conversation.