TheTorProject/lepidopter #79: run_oonideckgen is not run when network is up

sweemeng, 2016-09-15:
Currently there is a script that is not run when networking is up
```
/etc/network/if-up.d/run_oonideckgen
```
This script won't be run when network is up because it is not run by default without network manager. It can be fixed with adding the following at the eth0 section in /etc/network/interfaces.d/setup

```
post-up /etc/network/if-up.d/run_oonideckgen
```

Also script need to be executable

anadahz, 2016-09-15:
Network interface configuration tools for ifup and ifdown do not require network manager.

File `/etc/network/interfaces` includes the line: `source-directory /etc/network/interfaces.d` 
that sources files in `/etc/network/interfaces.d`.
For a file to be sourced there is no need to be executable.

The script `/etc/network/if-up.d/run_oonideckgen` is not required and will be removed in the newest release of lepidopter since ooniprobe daemon handles deck updates.

Thank you for reporting this.

anadahz, 2016-10-14:
I'm going to proceed by closing this issue, please feel free to re-open or submit a new issue.

closed 2016-10-14 by anadahz