TheTorProject/lepidopter #94: Set a PATH that will not be rewritten by any lepidopter updates

anadahz, 2017-01-06:
A PATH should be set where changes will not affected by any potential lepidopter updates.
As requested by paja on the ooni-talk mailing list.

hellais, 2017-01-06:
> I think in general it would be very useful if user can modify the URL list from the web interface or otherwise. It also increases the usability of OONI by a lot.

Yes I agree on this. I believe this feature should not be a lepidopter specific one, but it should rather be part of the ooniprobe distribution itself. We already support changes that persist across updates (for example settings and enabled decks) and this would just be one more of them.

Possibly related to this is also: https://github.com/TheTorProject/ooni-probe/issues/609

anadahz, 2017-05-05:
This has been implemented in: https://github.com/TheTorProject/lepidopter/releases/tag/v1.0.0

closed 2017-05-05 by anadahz