TheTorProject/lepidopter #95: Show simple webpage during update

ja-pa, 2017-01-09:
There is no indication of ongoing update through web interface. It would be nice to have some simple page which could highlight that something is going on.

darkk, 2017-01-09:
@ja-pa do you mean lepidopter software update/upgrade or update on network measurement progress?

ja-pa, 2017-01-10:
I mean lepidopter software update/upgrade.

muellermartin, 2019-07-12:
I've just installed Lepidoter on a Raspberry Pi B+ and I was wondering why I could not open the web interface although the device was reachable via network. Only after I used SSH to look at the list of processes (to look for something like a web server), I saw that a Python script was writing to `/var/log/ooni/lepidopter-update.log` which indicated that an update was running:

```
1970-01-01 00:01:18,666 - INFO - Checking for update
1970-01-01 00:01:19,932 - ERROR - Failed to learn the latest version
2019-07-12 15:27:15,969 - INFO - Checking for update
2019-07-12 15:27:18,064 - INFO - Updating 8->9
2019-07-12 15:27:34,695 - INFO - Running install script
```

The Raspberry Pi model that I'm using is quite slow, so Python modules which needs compiled parts are taking quite some time to install. I even tried to restart the device once before I noticed an ongoing update. Thus a webpage indicating that an update is in progress, would be really helpful.