TheTorProject/lepidopter #101: Change name to OONI Pi

elioqoshi, 2017-09-15:
Lepidopter is quite confusing as a name and not having the OONI brand as part of it, distances it from OONI. Calling it OONI Pi would make more sense as it has brand consistency, is a short name and sounds phonetically great (ooniepie)

What do you think?

hellais, 2017-09-16:
I think this is a solid idea and would also reduce confusion with the lepidopter name, that I have seen people have in remembering and pronouncing it.

elioqoshi, 2017-09-17:
@agrabeli @lorenzoPrimi @bassosimone Thoughts?

lorenzoPrimi, 2017-09-17:
OONI Pi seems too much short, but I agree with the renaming idea

hellais, 2017-09-17:
I am too to propose something radical: how about we just call it "OONI Probe". 

Effectively it's just a different target platform for "OONI Probe" and its actually mostly the same code base. It's not like on mobile we call the software "OONI Mobile" there.

hellais, 2017-09-17:
cc @anadahz

elioqoshi, 2017-09-17:
>OONI Pi seems too much short, but I agree with the renaming idea

I think the same, but phonetically it sounds great. OONI Octopi is another alternative.

elioqoshi, 2017-09-17:
@hellais That actually makes total sense. OONI Probe is available for Web and Mobile, while the RPi alternative has a different name. We should call it OONI Probe as that would make the most sense.

anadahz, 2017-09-17:
The idea of having the same name for all OONI programs/applications sounds good.

Lepidopter is not a program or application but a distribution based on Debian jessie with a number of programs and dependencies to assist ooni-probe the python program of OONI (https://github.com/TheTorProject/ooni-probe) to automatically perform daily measurements.

Lepidopter is currently available for a number of Raspberry Pi hardware revisions, with future releases targeting different platforms.

hellais, 2017-09-17:
@anadahz yes, you are right, however what the end users sees is just ooniprobe. Calling it anything else just leads to confusion.
Ooniprobe on mobile also includes a bunch of other libraries and stuff. The tests are actually implemented in measurement-kit.
The only reason why somebody downloads lepidopter is to run ooniprobe.

bassosimone, 2017-09-18:
Renaming it to OONI Probe would be misleading: we want the things we ship to have a minimal level of differentiation such that, when they report a bug, we understand what they're talking about.

Hence, given the choice between OONI Probe and OONI Pi, I have strong preference for the latter.

Since there is "OONI Probe Android", perhaps there could also be the "OONI Probe Pi". That is, a see a pattern "$maker $role $platform" in all the names proposed.

bassosimone, 2017-09-18:
To clarify, after a discussion on IRC with @hellais: I am saying basically that the repository should clearly contain "-pi" in the name and also the version string. Otherwise we create confusion into the users when it is time for them to report a bug (and this is something we would not like to happen).

elioqoshi, 2017-09-18:
If it's OONI Probe Web, OONi Probe Android, sure calling it OONI Probe Pi makes sense. But do we differentiate? I don't think we should differentiate because this is mostly a branding decision. Reporting bugs goes to a specific platform of Probe.

elioqoshi, 2017-09-18:
Completely agree that all repos should have the platform in their name.

elioqoshi, 2017-09-18:
I'd even go there and call it ooni-probe-raspberrypi or something to be very clear

bassosimone, 2017-09-18:
> Reporting bugs goes to a specific platform of Probe.

I don't think the renaming in itself would make the situation worst, but in general I would not assume this is something as easy as that. I've seen people that know very well our platform reporting bugs for features required in mobile devices onto the desktop repository. I believe that, probably, with my comment I raised a different problem that in itself has not that much to do with the naming.

bassosimone, 2017-09-18:
Okay, after some more discussion on IRC, I understand the rationale of renaming lepidopter for consistency even if we're not actively developing it at the moment. As I said on IRC to @elioqoshi, I believe that the goal of being able to say "OONI Probe is available on $platforms" is very good.

bassosimone, 2017-09-18:
More discussion points from IRC:

- we probably don't need to rename the internals of the software

- we should probably reference into the documentation that "OONI Probe for RaspberryPi" was "formerly known as Lepidopter", so to address potential confusion induced by the rename

anadahz, 2017-09-18:
As I mentioned in https://github.com/TheTorProject/lepidopter/issues/101#issuecomment-330079512 although lepidopter is currently targeted for Raspberry Pi future releases will support different hardware platforms  hence the name OONI Probe *Pi* will be misleading.

> The only reason why somebody downloads lepidopter is to run ooniprobe.

A public example of lepidopter usage apart from running ooniprobe is the [blocked project](https://blocked.org.uk).

bassosimone, 2017-09-18:
@anadahz could you elaborate on such plans?

hellais, 2017-09-18:
@anadahz yeah the proposal is not to call it "OONI Probe Pi", but just a"OONI Probe" as we advertise it as a way for people to run "OONI Probe".

Are you sure others are using it for anything other? If that is the case they are going to be very sad when the auto updates change their system.

In any case if there are to be other versions of it supporting other stuff (which are not available publicly at the moment) we should come up with some alternative branding for that.

bassosimone, 2017-09-18:
> In any case if there are to be other versions of it supporting other stuff (which are not available publicly at the moment) we should come up with some alternative branding for that.

👍

anadahz, 2017-09-18:
@bassosimone there are much cheaper embedded devices out there such as the OrangePi that can perform similarly well to the Raspberry PI.
In communication with some people of Devuan and looking at armbian release building for a number of targets with CI and automated builds seems like an approach that lepidopter development should follow.

anadahz, 2017-09-18:
> Are you sure others are using it for anything other? If that is the case they are going to be very sad when the auto updates change their system.

Indeed perhaps we should offer an option to disable automatic updates in lepidopter for users that would like to do their own administration and maintenance.

bassosimone, 2017-09-18:
@anadahz sure, there is opportunity, but are there concrete plans? I mean, is there a roadmap of something concrete enough to justify lepidopter as standalone platform?

agrabeli, 2017-09-18:
I think we are over-engineering this.

Lepidopter runs OONI Probe, therefore it *is* OONI Probe. As such, it only makes sense to call it OONI Probe.

Calling it anything else is not only confusing, but it is also extremely *misleading*. It implies that it is different software, when actually it is not. 

Whatever scripts exist around OONI Probe in "Lepidopter" are irrelevant to users. They care about OONI Probe, which is designed to measure censorship. They don't care about the scripts that make OONI Probe run on hardware platforms (or any other platform, for that matter).

hellais, 2017-09-18:
Ok, I am going to try to bring this discussion back into context a second.

**Some Facts**

1. Lepidopter, as in what users download from: https://ooni.torproject.org/install/, via the install guide https://ooni.torproject.org/install/lepidopter/, in particular the version https://get.ooni.torproject.org/lepidopter/lepidopter-v1.0.0-armel.img.zip, is primarily for running OONI Probe.

2. When people download and install lepidopter from the links in point 1, and the web interface via the instructions listed in the official installation guide (via `lepidopter.local`) see ONLY the OONI Probe web interface

3. If people use the version of lepidopter in point 1. for running their own custom software, their installation is quite likely going to break since we ship as part of it an auto updater system that touches also system files

4. We don't may any guarantee in any of the official lepidopter install guides that we support running custom experiments via lepdiopter and there is no clear plan for how we should support this or if we should support this.

**Some speculations**
(i.e. I have no hard evidence to support these claims, but based on interactions with people and some less strong evidence I find them quite probable)

1. 99.9% of the people that download and install lepidopter are downloading and installing it because they want to run "OONI Probe"

2. People that are using lepidopter for running their own custom tests are probably better off using something else that doesn't come with a auto-updater and want to do customisations of the image.

3. Most power users that want 2. actually end up making their own lepidopter look-alike (example: Venezuela Inteligiente) since the degree of customisation possible in lepidopter is not flexible enough and is error prone.

**What is the problem anyways?**

1. People are confused by the fact that to run OONI Probe on raspberry pi they need to download a thing that is called lepidopter

2. The name lepidopter itself is not the most simple. I have had countless conversations on the telephone with partners that went something like this:

Me: Now open your browser and connect to `lepidopter.local`
Them: Lepi-what?
Me: `lepidopter.local`, L-E-P-I-D-O-P-T-E-R
Them: It's doesn't work
Me: Can you spell it for me
Them: L-E-P-T-I-D-O-P-T
Me: Wait, there is a typo there, try this
Them: It still doesn't work
Me: `Goes through all the troubleshooting URLs in: https://ooni.torproject.org/install/lepidopter/#troubleshooting-service-discovery`

3. People find in general the name confusion and it's unclear the relation it has with OONI and OONI Probe. See: https://twitter.com/craftNeui/status/909425924172546048 and more.

4. It's important to build trust to consolidate all our projects under the OONI brand

**What is irrelevant/out of scope**

1. What Lepidopter could be in the distant future where it runs even on washing machines

2. What Lepidopter could be in the distant future where it adds support for netflix and home automation

3. The fact that people are using Lepidopter in ways that we don't officially support (ex. ORG for Blocked.org) and are risking breaking their fleet when we push an over-zealous update

4. What features we could be adding that would require N amount of development work to support 1-3.

**What matters**

The Lepidopter name is confusing. Users install Lepidopter to run OONI Probe, therefore the only reasonable thing is to call it OONI Probe.

bassosimone, 2017-09-18:
My opinions on this:

1. I think renaming the image distributed by OONI for OONI's sake is clearly a smart move in light of the issues that uses have had with the name. FTR, in the past we renamed `libight` to `measurement-kit` after @hellais and I noticed that people were not able to spell it correctly.

2. I think the repository name can even stay lepidopter (or perhaps `ooni-lepidopter` if we want to make it more clear the association with OONI): as @agrabeli says, it is not a user facing product, so it is unlikely that the name given to scripts will have an impact on users' understanding

3. it would be interesting to have the feature that we can configure the network name, because apparently typing `lepidopter` is a thing creating issues to users; I guess it should be configurable and that when building an image for OONI it should be configured to be `ooni.local`

elioqoshi, 2017-09-18:
Agreed with @agrabeli, we are bikeshedding over this (although I enjoyed the extensive analysis of @hellais ).

>I think the repository name can even stay lepidopter (or perhaps ooni-lepidopter if we want to make it more clear the association with OONI): as @agrabeli says, it is not a user facing product, so it is unlikely that the name given to scripts will have an impact on users' understanding

I think we should rename it sooner or later as it might be confusing for new contributors. Doesn't need to be instant though. 

So let's wrap it up and just incubate it within the OONI Probe sub-brand?

anadahz, 2017-09-18:
> @anadahz sure, there is opportunity, but are there concrete plans? I mean, is there a roadmap of something concrete enough to justify lepidopter as standalone platform?

@bassosimone lepidopter has been used as a standalone platform way before for a number of public and private projects. The first known official public release with the name lepidopter was referenced almost 4 years ago - December 2013. Time and availability define the progress and the milestones of the roadmap for the lepidopter distribution, but for sure is not a dead project.

anadahz, 2017-09-18:
> I think we are over-engineering this.

> Lepidopter runs OONI Probe, therefore it is OONI Probe. As such, it only makes sense to call it OONI Probe.

> Calling it anything else is not only confusing, but it is also extremely misleading. It implies that it is different software, when actually it is not.

> Whatever scripts exist around OONI Probe in "Lepidopter" are irrelevant to users. They care about OONI Probe, which is designed to measure censorship. They don't care about the scripts that make OONI Probe run on hardware platforms (or any other platform, for that matter).

@agrabeli

In a similar way perhaps we should name OONI Probe a system which uses iOS or *nix distribution s when it is solely used to run ooniprobe?

> Whatever scripts exist around OONI Probe in "Lepidopter" are irrelevant to users. They care about OONI Probe, which is designed to measure censorship. They don't care about the scripts that make OONI Probe run on hardware platforms (or any other platform, for that matter).

The same applies to any other distribution out there that supports running ooniprobe.

hellais, 2017-09-18:
> The first known official public release with the name lepidopter was referenced almost 4 years ago - December 2013.

It's unclear to me how that is even possible, given the fact that the first commit (of a 100 line shell script) is in March 2014: https://github.com/TheTorProject/lepidopter/commit/52cf01245e7e58f4ed407fd439b285225ef4a6cd.
Did you create time-travel?

anadahz, 2017-09-18:
> It's unclear to me how that is even possible, given the fact that the first commit (of a 100 line shell script) is in March 2014: 52cf012.
Did you create time-travel?

[Initial commit Lepidopter - anadahz committed on Dec 4, 2013](https://github.com/anadahz/lepidopter-spindle/commit/2797b99667993b7a2a49b71d674fb37da872adab)

hellais, 2017-09-18:
Anyways this thing is starting to take a ridiculous direction. Lepidopter IS OONI Probe. Lepidopter is not a platform. If you think otherwise, what would happen if we were to remove ooniprobe from lepidopter? What would be left? Debian....

I think I summed up all the reasoning in a fairly reasonable way in my comment above.

@anadahz you are the only one that is failing to see this and everybody else agrees.

I am going to say there is rough consensus to move forward with this renaming strategy and I suggest we do it together with the rest of the renamings and website revamping scheduled.

hellais, 2017-09-18:
Side comment. This project has been basically dormant for so much time and now we are spending an insane amount of time discussing a trivial thing as it's name.

[image]

Imagine if instead of spending all this time discussing this trivial matter, that everybody agrees on, we spent it actually making OONI Probe better?

Just a thought...

darkk, 2017-09-18:
> @anadahz you are the only one that is failing to see this and **everybody else agrees**.

HÄNDE HOCH! FORMAL LOGIC POLICE!

darkk, 2017-09-19:
My opinion on this flame-capable matter is:
- leave lepidopter as-is in the repo and code (to minimize renaming: both for reasons of respecting the names and reducing the labor and risks of typos)
- name the image _OONI Probe_ in OONI documentation AND mention that the platform to run _OONI Probe_ is _lepidopter_ that is _Debian GNU/Linux derivative_. _Tails_, _Whonix_ and _Astra-Linux_ deserve names, why _lepidopter_ does not? RIPE has _RIPE Atlas Probe_, but that's modified OpenWRT (and they acknowledge that in presentations: [1](https://www.ripe.net/about-us/press-centre/publications/presentations/2013/atlas-at-hackerspaces/at_download/download), [2](https://www.ripe.net/about-us/press-centre/publications/presentations/2016/ripe-atlas-a-measurement-network-cisco-iot-symposium/at_download/download))
- have BOTH http://ooni.local and http:/lepidopter.local URLs if possible, the first one is the most typo-friendly, the second is for compatibility. I would also consider redirecting from http://uni.local to http://ooni.local.

IMHO user will not be confused by name "lepidopter" that is mentioned in the documentation as the name of Debian derivative.

hellais, 2017-09-19:
I am sorry for having gotten a bit worked up in this ticket and having replied with a bit of an unfriendly tone 🙏 

I think that to some extent this got in the way of properly explaining what I was proposing.

I perfectly agree with what @darkk says in the above and I apologise if that was not clear in my previous responses.

The goal of this is just to reduce friction in how users learn about lepidopter and use it. That is to say:

* If that is desired we can keep this repo named this way
* We just update the installation pages for Lepidopter and where it says "Lepidopter" we say "OONI Probe" (https://ooni.torproject.org/install/lepidopter/ -> https://ooni.torproject.org/install/raspberrypi and `s/Lepidopter/OONI Probe/g`)
* If desired in the fore-mentioned install pages we add a note saying that OONI Probe for Raspberry Pi is powered by Lepidopter, or any other preferred wording
* The `.img` we distribute is renamed from `lepidopter-v1.0.0-armel.img.zip` to `ooniprobe-raspberrypi-v1.0.0-armel.img.zip` or something of the like
* ooniprobe.local etc., becomes the default hostname for accessing OONI Probe

Again sorry for not having been clearer.

@anadahz would this work for you?

adquadratum, 2018-01-16:
Elio I have some information to share about my install of Lepidopter in Canada.  What is the best way to do that?

 

Greg Rudy

 

Sent: Friday, September 15, 2017 at 12:59 PM

anadahz, 2018-01-18:
> Elio I have some information to share about my install of Lepidopter in Canada.  What is the best way to do that?

@adquadratum You could share this information by opening a new issue: https://github.com/TheTorProject/lepidopter/issues/new

If you would like to do this in private please use one of the contact methods mentioned in:
https://github.com/TheTorProject/lepidopter/issues/new

anadahz, 2018-12-02:
I'm going to close this issue as wontfix. 
The name of this distribution is around for so long to rename it.

closed 2018-12-02 by anadahz

elioqoshi, 2018-12-02:
You haven't addressed what @hellais asked you about and yet you move on and close the discussion? While the decision died down I don't think this is solely your decision @anadahz to take.

anadahz, 2018-12-02:
@elioqoshi thanks for your reply, please feel free to reopen this issue if there is any constructive feedback.