TheTorProject/lepidopter #106: ooniprobe is not running and web interface fails to load decks and recent results

muellermartin, 2019-07-12:
I've just installed Lepidopter 1.0.0 on a Raspberry Pi B+ and when I open the web interface, I get this error twice as popup:

> Failed to load decks
> SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data

The ooniprobe Dashboard shows that ooniprobe is not running and that it failed to load decks and recent results.

Using the refresh button on the top right results in the same error messages.

A quick look at the browser's network debugger shows that the requests to `/status`, `/deck`, `/measurement` and `/nettest` are failing with HTTP error code 500.

The response for `/status` shows following traceback as result:

<details>
    <summary>Click to expand</summary>

```
web.Server Traceback (most recent call last):
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
/usr/local/lib/python2.7/dist-packages/twisted/internet/defer.py:151 in maybeDeferred
150    try:
151        result = f(*args, **kw)
152    except:
/usr/local/lib/python2.7/dist-packages/klein/_app.py:134 in execute_endpoint
133        endpoint_f = self._endpoints[endpoint]
134        return endpoint_f(self._instance, *args, **kwargs)
135
/usr/local/lib/python2.7/dist-packages/klein/_app.py:250 in _f
249            def _f(instance, request, *a, **kw):
250                return _call(instance, f, request, *a, **kw)
251
/usr/local/lib/python2.7/dist-packages/klein/_app.py:55 in _call
54        args = (__klein_instance__,) + args
55    result = __klein_f__(*args, **kwargs)
56    if iscoroutine(result):
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:69 in wrapper
68                raise WebUIError(404, "Invalid XSRF token")
69            return f(instance, request, *a, **kw)
70
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:278 in api_status
277    def api_status(self, request):
278        return self.render_json(self.status, request)
279
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:260 in render_json
259        request.setHeader('Content-Type', 'application/json')
260        request.setHeader('Content-Length', len(json_string))
261        return json_string
/usr/local/lib/python2.7/dist-packages/twisted/web/http.py:1271 in setHeader
1270        """
1271        self.responseHeaders.setRawHeaders(name, [value])
1272
/usr/local/lib/python2.7/dist-packages/twisted/web/http_headers.py:220 in setRawHeaders
219        encodedValues = [_sanitizeLinearWhitespace(v)
220                         for v in self._encodeValues(values)]
221
/usr/local/lib/python2.7/dist-packages/twisted/web/http_headers.py:40 in _sanitizeLinearWhitespace
39    """
40    return b' '.join(headerComponent.splitlines())
41
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
```
</details>

And the other requests (`/deck`, `/measurement` and `/nettest`) show this traceback as result:

<details>
    <summary>Click to expand</summary>

```
web.Server Traceback (most recent call last):
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
/usr/local/lib/python2.7/dist-packages/twisted/internet/defer.py:151 in maybeDeferred
150    try:
151        result = f(*args, **kw)
152    except:
/usr/local/lib/python2.7/dist-packages/klein/_app.py:141 in execute_error_handler
140        """
141        return handler(self._instance, request, failure)
142
/usr/local/lib/python2.7/dist-packages/klein/_app.py:368 in _f
367            def _f(instance, request, failure):
368                return _call(instance, f, request, failure)
369
/usr/local/lib/python2.7/dist-packages/klein/_app.py:55 in _call
54        args = (__klein_instance__,) + args
55    result = __klein_f__(*args, **kwargs)
56    if iscoroutine(result):
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:69 in wrapper
68                raise WebUIError(404, "Invalid XSRF token")
69            return f(instance, request, *a, **kw)
70
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:255 in web_ui_error
254            "error_message": error.message
255        }, request)
256
/usr/local/lib/python2.7/dist-packages/ooniprobe-2.3.0-py2.7.egg/ooni/ui/web/server.py:260 in render_json
259        request.setHeader('Content-Type', 'application/json')
260        request.setHeader('Content-Length', len(json_string))
261        return json_string
/usr/local/lib/python2.7/dist-packages/twisted/web/http.py:1271 in setHeader
1270        """
1271        self.responseHeaders.setRawHeaders(name, [value])
1272
/usr/local/lib/python2.7/dist-packages/twisted/web/http_headers.py:220 in setRawHeaders
219        encodedValues = [_sanitizeLinearWhitespace(v)
220                         for v in self._encodeValues(values)]
221
/usr/local/lib/python2.7/dist-packages/twisted/web/http_headers.py:40 in _sanitizeLinearWhitespace
39    """
40    return b' '.join(headerComponent.splitlines())
41
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
```
</details>

anadahz, 2019-07-14:
Thank you for your reporting this issue.

Did you check if ooniprobe agent is running (`systemctl status ooniprobe`)?

Please note that lepidopter is based on the python version of ooniprobe that is now considered legacy, but not deprecated as there no replacement yet. Feel free to check out the [next generation OONI Probe CLI](https://github.com/ooni/probe-cli) and compile it for the Raspberry Pi.

muellermartin, 2019-07-15:
ooniprobe seems to be running. At least `systemctl status ooniprobe` shows `Active: active (running)` state and `ps aux | grep [o]oniprobe` returns a process: `/usr/bin/python /usr/local/bin/ooniprobe-agent start`.

anadahz, 2019-07-15:
Can you please try using a different browser with/or a clean profile and no plugins?

muellermartin, 2019-07-15:
Good point, I admit that I did not rule this out. But now I've tried to load the website additionally in Firefox Nightly, Safari and Google Chrome – sadly with the same result.

anadahz, 2019-07-15:
Can you run a test manually from cli?
Example:
`root@lepidopter:~# ooniprobe -v manipulation/http_invalid_request_line`

Perhaps you can also try to restart ooniprobe service.

muellermartin, 2019-07-15:
I've ran the test as user `lepidopter` with following output (seems to be successfull):

<details>
<summary>Click to expand</summary>

```
/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
  utils.CryptographyDeprecationWarning
It looks like this is the first time you are running ooniprobe
Please take a minute to read through the informed consent documentation and understand what are the risks associated with running ooniprobe.
Press enter to continue...

Type "yes" if you are fully aware of the risks associated with using ooniprobe and you wish to proceed
> yes

Now help us configure some things!
Should we upload measurements to a collector? (Y/n) 
Should we include your IP in measurements? (y/N) y
Should we include your ASN (your network) in measurements? (Y/n)  
Should we include your Country in measurements? (Y/n) 
How would you like reports to be uploaded? (onion, https, cloudfront) https
ooniprobe is now initialized. You can begin using it!
No test deck detected
Checking if backend is present
Tor is not running. Skipping IP lookup via Tor.
Looking up your IP address via ubuntu
Found your IP via a GeoIP service
Running task update-inputs
Detected first run
Updating the global inputs and resources
Moving /home/lepidopter/.ooni/resources/H6td6XbNaYCAmnNtmanifest.json to /home/lepidopter/.ooni/resources/manifest.json
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/1WHFpCr-8ArgvW5e00-LEGEND-category_codes.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/00-LEGEND-category_codes.csv
Moving /home/lepidopter/.ooni/resources/tor-bridges/mwtOoosyybo_rh6ytor-bridges-ip-port.csv to /home/lepidopter/.ooni/resources/tor-bridges/tor-bridges-ip-port.csv
Moving /home/lepidopter/.ooni/resources/maxmind-geoip/ir7psVp-uS0b3arGGeoIPASNum.dat to /home/lepidopter/.ooni/resources/maxmind-geoip/GeoIPASNum.dat
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/EfTQ6y7sMu-isuauglobal.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/global.csv
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/zclR2823ylcS78escis.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/cis.csv
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/Ds0h9_zfB6-H53Yy00-LEGEND-new_category_codes.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/00-LEGEND-new_category_codes.csv
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/ieyKc_pOShucDIEC00-LEGEND-country_codes.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/00-LEGEND-country_codes.csv
Moving /home/lepidopter/.ooni/resources/maxmind-geoip/vHk08OQxNx19sVHpGeoIP.dat to /home/lepidopter/.ooni/resources/maxmind-geoip/GeoIP.dat
Updating the inputs
Updating the inputs for country DE
Moving /home/lepidopter/.ooni/resources/citizenlab-test-lists/Uszpt0HLJTMaFMXDde.csv to /home/lepidopter/.ooni/resources/citizenlab-test-lists/de.csv
Looking up collector and test helpers with https://bouncer.ooni.io
Querying backend https://bouncer.ooni.io/bouncer/net-tests with {'net-tests': [{'test-helpers': ['tcp-echo'], 'version': '0.2', 'name': 'http_invalid_request_line', 'input-hashes': []}]}
Querying backend https://c.collector.ooni.io:443/invalidpath with None
Got this backend error message {u'error': 404}
Setting collector and test helpers for http_invalid_request_line
Using collector <ooni.backend_client.CollectorClient object at 0xb4cecfb0>
Starting dcc11fe6428efb1e (user-run)
Creating report with OONIB Reporter. Please be patient.
This may take up to 1-2 minutes...
Querying backend https://c.collector.ooni.io:443/report with {'data_format_version': '0.2.0', 'software_name': 'ooniprobe', 'test_version': '0.2', 'software_version': '2.3.0', 'test_name': 'http_invalid_request_line', 'test_start_time': '2019-07-15 13:35:57', 'format': 'json', 'input_hashes': [], 'probe_asn': 'AS680', 'probe_cc': 'DE'}
Created report with id 20190715T133558Z_AS680_ZI4YiCiwqRDqtndiFalaRosy8PkQvF6mliHhXUzxlYmOrlvdmi
Starting this task <generator object generateMeasurements at 0xb52fb170>
Running <<class 'ooni.nettest.NetTestCaseWithLocalOptions'> inputs=[None]> test_random_invalid_method
Connecting to 37.218.247.110:80
Running <<class 'ooni.nettest.NetTestCaseWithLocalOptions'> inputs=[None]> test_random_invalid_version_number
Connecting to 37.218.247.110:80
Running <<class 'ooni.nettest.NetTestCaseWithLocalOptions'> inputs=[None]> test_random_invalid_field_count
Connecting to 37.218.247.110:80
Running <<class 'ooni.nettest.NetTestCaseWithLocalOptions'> inputs=[None]> test_random_big_request_method
Connecting to 37.218.247.110:80
Connected to 37.218.247.110:80
Connected to 37.218.247.110:80
Connected to 37.218.247.110:80
Connected to 37.218.247.110:80
Closing connection
Checking if  == ZU2C / HTTP/1.1\n\r
random_invalid_method: Detected manipulation!

[... 36 lines cut ...]
------
2/4 completed
50.0% (ETA: 1s)
Checking all tasks for completion 2 == 4

Status
------
3/4 completed
75.0% (ETA: 0s)
Checking all tasks for completion 3 == 4

Status
------
4/4 completed
99.0% (ETA: 1s)
Checking all tasks for completion 4 == 4
Report ID: 20190715T133558Z_AS680_ZI4YiCiwqRDqtndiFalaRosy8PkQvF6mliHhXUzxlYmOrlvdmi
Closing report with id 20190715T133558Z_AS680_ZI4YiCiwqRDqtndiFalaRosy8PkQvF6mliHhXUzxlYmOrlvdmi
Querying backend https://c.collector.ooni.io:443/report/20190715T133558Z_AS680_ZI4YiCiwqRDqtndiFalaRosy8PkQvF6mliHhXUzxlYmOrlvdmi/close with None
Successfully performed report <ooni.tasks.ReportEntry object at 0xb4d19cf0>
None
Deleting log file
Finished dcc11fe6428efb1e (user-run)
```
</details>

anadahz, 2019-07-15:
It seems that ooniprobe runs OK (though with a different, note that some tests require extra privileges).
It may be related to this issue: https://github.com/scrapy/scrapyd/issues/311

Could you please check which version of Twisted are you running (`pip list |grep Twisted`)?

muellermartin, 2019-07-15:
```
lepidopter@lepidopter:~$ pip list |grep Twisted
/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
  utils.CryptographyDeprecationWarning
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Twisted (19.2.1)
```

anadahz, 2019-07-15:
Can you try to downgrade the Twisted pip package to an earlier version and restart ooniprobe service?
`sudo pip install == 'Twisted==18.7.0' && sudo systemctl restart ooniprobe`

muellermartin, 2019-07-15:
I had to remove the double equal before Twisted to make the command work (`sudo pip install 'Twisted==18.7.0' && sudo systemctl restart ooniprobe`).

After downgrading from Twisted 19.2.1 to 18.7.0, the dashboard showed the welcome screen with the setup and now everythings is reported to work as expected. I guess this issue is resolved – thanks a lot for your help!

anadahz, 2019-07-15:
I'm going to close this issue, thank you for reporting back.

closed 2019-07-15 by anadahz